ShadowBlip/InputPlumber #86: The back buttons are defined as L3 and R3 in ROGALLy.

kzkzkzkzkzk, 2024-05-07:
Hello. I am using inputPlumber in rogally, but the back button is defined as R3.L3. Is it possible to change it?

ShadowApex, 2024-05-07:
We can change the default profile so these are not mapped to R3/L3 (#88).

You can change the default profile to map those buttons to any input you'd like. The default profile is located at: `/usr/share/inputplumber/profiles/default.yaml`.

Alternatively, you could also create new profiles and use the `busctl` command to switch to the profiles you've made. Just change `/path/to/your/profile.yaml` to the location of the profile you made:

```bash
busctl call org.shadowblip.InputPlumber /org/shadowblip/InputPlumber/CompositeDevice0 org.shadowblip.Input.CompositeDevice LoadProfilePath s /path/to/your/profile.yaml
```

Creating custom profiles and switching to them will be made easier once InputPlumber is integrated into different UI's like [OpenGamepadUI](https://github.com/ShadowBlip/OpenGamepadUI).

kzkzkzkzkzk, 2024-05-07:
thank you!
I was able to change the default settings and change them to any button. But in the case of DS5-edge, how can I define the L4.R4 buttons to be true? I want to define the ROGALLY back button as the L4.R4 button that can be reassigned for each game, but no matter how many times I try, I can't enable it.

ShadowApex, 2024-05-07:
If you just remove `LeftPaddle` and `RightPaddle` mappings from the profile, they should default to the DualSense L4/R4 buttons.

kzkzkzkzkzk, 2024-05-07:
If it's default, is it emulating an xbox controller on the DS5edge? Is the profile for ALLY being used? It wasn't recognized even when I deleted it. Does this change depending on the MCU or bios?

ShadowApex, 2024-05-07:
The default device to emulate on the ROG Ally is currently the DualSense Edge controller:

https://github.com/ShadowBlip/InputPlumber/blob/main/rootfs/usr/share/inputplumber/devices/50-rog_ally.yaml#L39-L43

You can also override the gamepad to emulate in the profile you create.

kzkzkzkzkzk, 2024-05-07:
https://github.com/ShadowBlip/InputPlumber/blob/main/rootfs/usr/share/inputplumber/capability_maps/ally_type1.yaml

Isn't this what it's used for?

PR #88 (2024-05-08): fix(Default Profile): don't map back paddles to stick buttons by default
Fixes #86
closed 2024-05-08 by ShadowApex