ShadowBlip/InputPlumber #134: [Regression] 0.25.0 can't use D-pad on steam client

philmmanjaro, 2024-06-28:
Seems with 0.25.0 we find a regression that d-pad is not working at all. It works in OGUI overlay, however not in Steam UI. 0.25.1 has the same effect. Tested on OrangePi Neo in Xbox and DSE mode.

pastaq, 2024-06-28:
I can't reproduce this

philmmanjaro, 2024-06-28:
Currently testing if 89ee1ae427411e155c9153aef1d651f476b0b3a4 created the regression. 0.24.2 worked with the same setup.

philmmanjaro, 2024-06-28:
Reverting makes it work again ...

pastaq, 2024-06-28:
Do you have a profile you are loading that you have modified? I still can't reproduce this on any session.

philmmanjaro, 2024-06-29:
The only change we do on our end is this:

	sed -i -e 's/- ds5/- gamepad/g' ${pkgdir}/usr/share/inputplumber/devices/50-orangepi_neo.yaml
	sed -i -e 's/- ds5-edge/- gamepad/g' ${pkgdir}/usr/share/inputplumber/devices/50-rog_ally.yaml

When I revert the commit it works on the 8840u unit. Didn't had time yet to cross check the ALLY.

pastaq, 2024-06-29:
Fixed in #135

closed 2024-06-29 by pastaq