ShadowBlip/InputPlumber #193: Keyboard shortcut chord has changed

uejji, 2024-08-19:
It seems that Steam has changed the shortcut chord to bring up the on screen keyboard with an xbox controller to Guide+Select/Back/View.  However, InputPlumber is still sending the old chord Guide+X.

philmmanjaro, 2024-08-19:
Well, there are two profiles now. Xbox and Xbox Elite. We will check.

pastaq, 2024-08-19:
This is a bug in the steam client introduced when they tried to unify the big picture and steam deck mode chords. It wasn't an intentional change. This only affects Xbox controllers. Submit a bug report to Valve so they can change it back. We can't really support two separate chords for the same capability.

uejji, 2024-08-19:
Okay.  Since it's not an InputPlumber bug, I'm closing the issue.

closed 2024-08-19 by uejji