ShadowBlip/InputPlumber #202: Feature: Support running InputPlumber unprivileged

R1kaB3rN, 2024-09-14:
### Description

I'm interested in making InputPlumber available for clients through umu-launcher to fill in the gap of users unable to have their game controllers work, with the goal of the service being started by the client or umu-launcher at game launch in an unprivileged context.

### Problem

InputPlumber requires to be started and enabled as a privileged user to fulfill all of its tasks properly, making it both inaccessible for Flatpak applications and inconvenient as it would require clients to prompt their users for elevated privileges in the non-Flatpak case. Attempting to execute the `inputplumber` binary unprivileged results in:
```

[2024-09-14T19:03:19Z INFO  inputplumber] Starting InputPlumber v0.35.2
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(25))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(27))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(29))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(31))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(33))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(35))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(37))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(39))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(41))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(43))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(45))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(47))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(49))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(51))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(53))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(55))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(57))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(59))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(61))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(63))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(65))
[2024-09-14T19:03:19Z ERROR inputplumber::input::manager] Failed to run task to list device configs: JoinError::Cancelled(Id(67))
```

Due to this limitation, clients cannot simply run the binary like gamescope-dbus. As a result, users of popular launchers (e.g., Heroic Games Launcher) continue to depend on other clients or use other means to have control over their input devices (e.g., Steam Deck users adding a client app as a non-Steam game in Steam. In other cases, users may revert to using an old version of Wine due to their input devices being incorrectly mapped when running games through Proton.

Being able to run InputPlumber as an unprivileged user would avoid any manual intervention from the user or dependency on external apps to control their input devices.

### Proposal
Support running InputPlumber as an unprivileged user or at least a subset of InputPlumber's feature set in that case. After bringing this up to @ShadowApex, they said it should _theoretically_ be possible as long as there are polkits in place. However,  the policies would need to be created and installed as the root user, making this only doable for distribution packages. I'm wondering if there's another way or if this feature is just impossible to support given the current design? 

Any thoughts/suggestions on this are appreciated.

ShadowApex, 2024-09-16:
To expand on this, these are the different patterns we know about to hide the source input device from applications:

- Write udev rule to `/run/udev` to block source input device **(requires root)**. As mentioned, this is what we currently do in InputPlumber.
- Move input dev nodes (e.g. `/dev/input/event0`) to another path to prevent applications from finding the device **(requires root)**. This is what we _used_ to do, but in some cases, applications would open the file descriptor before we could actually move the device, causing the app to read inputs from both the source device and the virtual input device.
- Set the [SDL_HINT_GAMECONTROLLER_IGNORE_DEVICES](https://wiki.libsdl.org/SDL2/SDL_HINT_GAMECONTROLLER_IGNORE_DEVICES) and/or [SDL_HINT_HIDAPI_IGNORE_DEVICES](https://wiki.libsdl.org/SDL3/SDL_HINT_HIDAPI_IGNORE_DEVICES) environment variables when launching games to inform SDL to ignore the physical source devices and only use the virtual input devices **(no root required)**. This is what Steam Input does, but this technique only works with games that are using SDL to manage input devices. It also requires coordination with whatever launcher is being used (i.e. Heroic Games Launcher, OpenGamepadUI, Lutris, etc.). We've tried this technique with some success, but there's been problems, especially for native games that do not use SDL to manage input devices.
- Launch the game using `bwrap` (or similar sandbox) and only expose the virtual input devices that InputPlumber creates **(no root required)**. This would theoretically work with all input implementations, but would also require launcher coordination so games are launched in a sandbox, and a secondary process would be responsible for exposing the virtual devices to the game's sandbox. I have tried doing this in the past, but could never get it to work correctly. My past attempts were trying to use `bwrap` to create a fake `/dev` directory that we could bind mount into the sandbox to selectively expose input devices, but I think the games just wouldn't see any input devices.

We're open to other ideas if there's another technique we could use. Not the best solution, but one option could be to check if the InputPlumber service is available, and if not, the launcher could recommend to the user that they install InputPlumber through their package manager (or systemd extension).

braiam, 2024-10-23:
Would a patch to upstream kernel be a solution to support this use case?

strycore, 2025-01-21:
> One possibility might be to run a second nested jail inside the flatpak jail to filter the device nodes to only the ones provided by InputPlumber. I have no idea how difficult that would be or if it is even possible since running the jail also requires SUID, at least for firejail. I don't remember if bwrap uses SUID to start a jail. 

> Launch the game using bwrap (or similar sandbox) and only expose the virtual input devices that InputPlumber creates (no root required). This would theoretically work with all input implementations, [...]

Since umu is essentially a way to run games in a Pressure Vessel environment, this seems the easiest way to implement this feature

pastaq, 2025-01-21:
Can a jailed application use dbus?
If so then we might be able to hook integration into client applications inside the jail. 
- Client app reads all CompositeDevice instances and gets a list of their target devices.
- Client app runs the application, explosing only those target devices.

If dbus isn't available then we could use InputPlumber to add a udev property when targets are created that can be provbed by the client to know it's an InputPlumber target device.

Both of these approaches might be limited in the same way. If we're not able to adjust the list of devices after the jail is started then the targets will need to be known and running before the jail is started, and changing controller types will require a restart of the game.

Pandapip1, 2025-07-28:
I'm not too familiar with it, but I think you could use eBPF to intercept syscalls to read directories and open files and cause them to fail?

TheRealYasri, 2026-02-22:
Not sure if this is breaking flatpak rules or anything. What I have been doing for games launched by faugus launcher is have it make a desktop icon. Then edit it to this. the profile yaml file for that game is in its wine prefix root folder.

**Edited one**
[image]
**Original one**
[image]