ShadowBlip/InputPlumber #208: Flydigi Apex 4 support

zany130, 2024-09-30:
As previously discussed on discord, this is to track adding support to the [Flydigi Apex 4 controller](https://www.amazon.com/PC-Controller-Feedback-Force-Adjustable-Full-Color-Vibration/dp/B0CRR1WR9W)

Relevant discussions and dumps can be found here:

https://github.com/hhd-dev/hhd/issues/73

https://github.com/libsdl-org/SDL/issues/10161

https://github.com/zany130/hwinfo/tree/master/devices/Apex4

Fabianoshz, 2024-12-02:
@zany130 I don't know if this helps you but I've managed to make the back buttons of the flydigi work with a set of patches:

- First I've patched xpad as described [here](https://github.com/paroj/xpad/pull/268) with this code:
```
diff --git a/xpad.c b/xpad.c
index d5a016e..442809e 100644
--- a/xpad.c
+++ b/xpad.c
@@ -95,6 +95,7 @@
 #define MAP_SELECT_BUTTON		(1 << 3)
 #define MAP_PADDLES			(1 << 4)
 #define MAP_PROFILE_BUTTON		(1 << 5)
+#define MAP_FLYDIGI_BUTTONS		(1 << 6)
 
 #define DANCEPAD_MAP_CONFIG	(MAP_DPAD_TO_BUTTONS |			\
 				MAP_TRIGGERS_TO_BUTTONS | MAP_STICKS_TO_NULL)
@@ -421,6 +422,19 @@ static const struct xpad_device {
 	{ 0x0000, 0x0000, "Generic X-Box pad", 0, XTYPE_UNKNOWN }
 };
 
+// A "flavor" is an aftermarket variant of an existing model supporting
+// additional features.
+static const struct xpad_flavor {
+	u16 idVendor;
+	u16 idProduct;
+	char *product;
+	u8 mapping;
+} xpad_flavor[] = {
+	{ 0x045e, 0x028e, "Flydigi VADER3", MAP_PADDLES | MAP_FLYDIGI_BUTTONS },
+	{ 0x045e, 0x028e, "Flydigi VADER4", MAP_PADDLES | MAP_FLYDIGI_BUTTONS },
+	{ 0x0000, 0x0000, NULL, 0 }
+};
+
 /* buttons shared with xbox and xbox360 */
 static const signed short xpad_common_btn[] = {
 	BTN_A, BTN_B, BTN_X, BTN_Y,			/* "analog" buttons */
@@ -428,6 +442,13 @@ static const signed short xpad_common_btn[] = {
 	-1						/* terminating entry */
 };
 
+/* used for extra buttons in addition to paddles on Flydigi Vader Pro 3*/
+static const signed short xpad_btn_extra[] = {
+	BTN_TRIGGER_HAPPY9, BTN_TRIGGER_HAPPY10, /* C, Z face buttons */
+	BTN_TRIGGER_HAPPY11,			/* circle */
+	-1						/* terminating entry */
+};
+
 /* original xbox controllers only */
 static const signed short xpad_btn[] = {
 	BTN_C, BTN_Z,		/* "analog" buttons */
@@ -1012,6 +1033,17 @@ static void xpad360_process_packet(struct usb_xpad *xpad, struct input_dev *dev,
 		input_report_abs(dev, ABS_RZ, data[5]);
 	}
 
+	/* Additional buttons for Flydigi Vader Pro 3 presenting as 360 pad. */
+	if (xpad->mapping & MAP_FLYDIGI_BUTTONS) {
+		input_report_key(dev, BTN_TRIGGER_HAPPY9, data[19] & BIT(0));   // C
+		input_report_key(dev, BTN_TRIGGER_HAPPY10, data[19] & BIT(1));  // Z
+		input_report_key(dev, BTN_TRIGGER_HAPPY5, data[19] & BIT(3));   // Leftmost paddle (M2)
+		input_report_key(dev, BTN_TRIGGER_HAPPY6, data[19] & BIT(5));   // Second to leftmost (M4)
+		input_report_key(dev, BTN_TRIGGER_HAPPY7, data[19] & BIT(4));   // Second to rightmost (M3)
+		input_report_key(dev, BTN_TRIGGER_HAPPY8, data[19] & BIT(2));   // Rightmost paddle (M1)
+		input_report_key(dev, BTN_TRIGGER_HAPPY11, data[20] & BIT(0));  // Circle
+	}
+
 	input_sync(dev);
 
 	/* XBOX360W controllers can't be turned off without driver assistance */
@@ -2246,6 +2278,13 @@ static int xpad_init_input(struct usb_xpad *xpad)
 			input_set_capability(input_dev, EV_KEY, xpad_btn_paddles[i]);
[... 96 lines cut ...]

options:
  auto_manage: true
```

```
# yaml-language-server: $schema=https://raw.githubusercontent.com/ShadowBlip/InputPlumber/main/rootfs/usr/share/inputplumber/schema/capability_map_v1.json
# Schema version number
version: 1

# The type of configuration schema
kind: CapabilityMap

# Name for the device event map
name: flydigi-vader-4-pro

# Unique identifier of the capability mapping
id: flydigi-vader-4-pro

# List of mapped events that are activated by a specific set of activation keys.
mapping:

# List of events to filter from the source devices
filtered_events: []
```

zany130, 2024-12-02:
Oh, interesting! I'll try this out! So, basically, this is patching the pad to support the back buttons of the Apex 4 when in input mode and then using input plumber to emulate an elite controller so it shows in Steam?

The only thing missing with this would be gyro and the extra "circle button" (button next to home)

zany130, 2024-12-02:
No the circle button doesn't send anything in x-input mode

The rest is all d-input mode

In d-input it sends the right trigger so it might not be possible to map it separately from the the right trigger thinking back on it

Gyro is sent in two hidraws.

The flydigi fly mouse (gyro mouse actived by the circle button (in d-input it sends both the right trigger to the OS and internally it also activates the fly mouse) it sent in a separate hidraw from everything else . There are also mouse buttons on this hidraw mapped to two of the back buttons (I think it was the inner two)

There is also what seems to be raw gyro data on the main hidraw were all the other buttons and sticks are , but a gyro mapping (button combo to enable gyro aim in the windows software) has to be set in the software for it to show up. What buttons are mapped do not matter all that matters is that some type of mapping is set.

zany130, 2024-12-05:
if it heelps this is how the controller is mapped in dinput

```
Input device ID: bus 0x3 vendor 0x4b4 product 0x2412 version 0x111
Input device name: "Flydigi Flydigi VADER3"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH) A
    Event code 305 (BTN_EAST) B
    Event code 306 (BTN_C) not used
    Event code 307 (BTN_NORTH) X
    Event code 308 (BTN_WEST) Y
    Event code 309 (BTN_Z) not used
    Event code 310 (BTN_TL) LB
    Event code 311 (BTN_TR) RB
    Event code 312 (BTN_TL2) LT
    Event code 313 (BTN_TR2) RT
    Event code 314 (BTN_SELECT) Select
    Event code 315 (BTN_START) Start
    Event code 317 (BTN_THUMBL) L3
    Event code 318 (BTN_THUMBR) R3
    Event code 319 (?)  M1
    Event code 398 (KEY_RED) Guide/home 
    Event code 704 (BTN_TRIGGER_HAPPY1) M2
    Event code 705 (BTN_TRIGGER_HAPPY2) M3
    Event code 706 (BTN_TRIGGER_HAPPY3) M4
    Event code 707 (BTN_TRIGGER_HAPPY4) not used
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)  Left stick X axis
      Value    127
      Min        0
      Max      255
      Flat      15
    Event code 1 (ABS_Y) Left stick Y axis
      Value    127
      Min        0
      Max      255
      Flat      15
    Event code 2 (ABS_Z)  Right stick Y axis
      Value    127
      Min        0
      Max      255
      Flat      15
    Event code 5 (ABS_RZ) Right stick X axis
      Value    127
      Min        0
      Max      255
      Flat      15
    Event code 9 (ABS_GAS) Right trigger and Circle button (sends a value of 2)
      Value      0
      Min        0
      Max      255
      Flat      15
    Event code 10 (ABS_BRAKE) Left trigger
      Value      0
      Min        0
      Max      255
      Flat      15
    Event code 16 (ABS_HAT0X) Dpad left + right
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y) Dpad Up + down
      Value      0
      Min       -1
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
    ```

belozer, 2025-03-04:
> [@zany130](https://github.com/zany130) I don't know if this helps you but I've managed to make the back buttons of the flydigi work with a set of patches:
>
> * First I've patched xpad as described [...]

Hi @Fabianoshz.  I'm new to using InputPlumber, so I need some help. Where I need place config and how apply it? Sorry for the stupid question.

Also about apply patches for xpad. That [PR is updated](https://github.com/paroj/xpad/pull/268) many times. Latest commit is 6b93b7e6b4473aadab0de870bdec91e8ee74de81 Your patch is actual?

Fabianoshz, 2025-03-04:
> Hi [@Fabianoshz](https://github.com/Fabianoshz). I'm new to using InputPlumber, so I need some help. Where I need place config and how apply it? Sorry for the stupid question.
>
> Also about apply [...]

About inputplumber things have changed since I've added my comment, you no longer need to add any patches to inputplumber, but your capability map should look like this:
```
# yaml-language-server: $schema=https://raw.githubusercontent.com/ShadowBlip/InputPlumber/main/rootfs/usr/share/inputplumber/schema/capability_map_v1.json                                                                                           
# Schema version number                                                                                                                                                                                                                              
version: 1                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                     
# The type of configuration schema
kind: CapabilityMap

# Name for the device event map
name: flydigi-vader-4-pro

# Unique identifier of the capability mapping
id: flydigi-vader-4-pro

# List of mapped events that are activated by a specific set of activation keys.
mapping:
  - name: RightPaddle1
    source_events:
      - gamepad:
          button: RightPaddle1
    target_event:
      gamepad:
        button: LeftPaddle2
  - name: RightPaddle2
    source_events:
      - gamepad:
          button: RightPaddle2
    target_event:
      gamepad:
        button: LeftPaddle1
  - name: RightPaddle3
    source_events:
      - gamepad:
          button: RightPaddle3
    target_event:
      gamepad:
        button: LeftPaddle3

  - name: LeftPaddle1
    source_events:
      - gamepad:
          button: LeftPaddle1
    target_event:
      gamepad:
        button: RightPaddle1
  - name: LeftPaddle2
    source_events:
      - gamepad:
          button: LeftPaddle2
    target_event:
      gamepad:
        button: RightPaddle2
  - name: LeftPaddle3
    source_events:
      - gamepad:
          button: LeftPaddle3
    target_event:
      gamepad:
        button: RightPaddle3

# List of events to filter from the source devices
filtered_events: []

```
Which I believe you should add to: `/usr/share/inputplumber/capability_maps/flydigi-vader-4-pro.yaml`.

Nothing should change for the `device` config:
```
# yaml-language-server: $schema=https://raw.githubusercontent.com/ShadowBlip/InputPlumber/main/rootfs/usr/share/inputplumber/schema/composite_device_v1.json
# Schema version number
version: 1

# The type of configuration schema
kind: CompositeDevice

# Name of the composite device mapping
name: flydigi-vader-4-pro

# Only use this profile if *any* of the given matches matches. If this list is
# empty,then the source devices will *always* be checked.
# /sys/class/dmi/id/product_name
matches: []

# Only allow a single source device per composite device of this type.
single_source: true

# One or more source devices to combine into a single virtual device. The events
# from these devices will be watched and translated according to the key map.
source_devices:
  - group: gamepad
    unique: true
    evdev:
      vendor_id: "045e"
      product_id: "028e"
      handler: event*

# The target input device(s) that the virtual device profile can use
target_devices:
  - xbox-elite

capability_map_id: flydigi-vader-4-pro

options:
  auto_manage: true
```

Which I believe you should add to: `/usr/share/inputplumber/devices/40-flydigi-vader-4-pro.yaml`.

I'm using an old version of xpad, but I would expect the latest patch should work just fine, which distro are you using? Instructions might differ based on what system you're using.

belozer, 2025-03-04:
@Fabianoshz thanks. I managed to introduce myself to the Xbox Elite controller on Steam. However, Steam hasn't yet been able to fix the paddle buttons, and the remapping might not have been finished somewhere. But that's a different problem (will check xpad). The most important thing is that I was able to introduce myself as a new controller. That was cool.

pastaq, 2025-03-05:
> About inputplumber things have changed since I've added my comment, you no longer need to add any patches to inputplumber, but your capability map should look like this:

These are great. Please PR to add them, then we can close this issue and this controller will work OOTB.

belozer, 2025-03-15:
I noticed one feature. If the Apex 4 is connected by wire, its Product ID is correctly determined. If connected via Dongle, then the gamepad is defined as Vader 3. 

It looks like a flaw on the part of the dongle firmware, I sent an email to Flydigi support asking them to put this situation in order.

They said that they had passed the information on to the developers and that they needed to wait for the latest news. I hope they will finalize the firmware fix and it will be much easier to distinguish Vader 3 from Apex 4.

belozer, 2025-06-10:
Hmm. Looks like we have good news from Steam.

https://store.steampowered.com/news/group/4397053/view/545611272206420771?l=russian

fenuks, 2025-07-27:
Perhaps they only create virtual DualSense controller, so games know they can use adaptive triggers, and in Apex 4 native DInput support, adaptive triggers use the same events as DualSense? If I knew how to do this in Linux, I could try it.

> Is there any way to tell in game that the adaptive triggers are active aside from feeling them?

I think they should be activated when you equip a weapon, since most games use them to imitate resistance of physical trigger in guns, but I didn't try yet to run any game in virtual machine since 3D performance is very bad.

> I can probably do part of that myself if you're not comfortable.

I don't have any Windows computer, only virtual machine, so if you could do that, I would be grateful. I could try to do that as well, but I would need some help with it. I presume you use Wireshark to capture USB communication?

pastaq, 2025-07-28:
Since we can already emulate a dualsense controller we should be getting the event when it happens inside InputPlumber already. We just need to identify the event, understand it, then figure out how to upload it to the FlyDigi controller.