ShadowBlip/InputPlumber #217: Ayaneo air 1s Gyro flipped

dansl, 2024-10-16:
When using a configuration that enables Gyro features (steam deck or DS-edge), the pitch and yaw are flipped.

uejji, 2024-10-16:
Hello again.

Can you try adding a mount_matrix to your config file?  See https://github.com/ShadowBlip/InputPlumber/blob/6d2c65fb0c66bcdf79e376232fa28c4b9aedbf0d/rootfs/usr/share/inputplumber/devices/50-ayaneo_air_plus.yaml#L53 as an example.

dansl, 2024-10-16:
That fixed it! 

But... Only fixed it for games that have in-game gyro support (ex: Horizon Zero Dawn). I notice that when I look at the Steam calibration test page Y and Z are flipped...

uejji, 2024-10-16:
You'll probably need to adjust the signs and value positions of the mount_matrix until you find the right combination.  Just make sure each row has only one 1/-1 and two 0s, and that each row has the 1/-1 in a different column.  Unfortunately I can't provide much more specific help than that.

dansl, 2024-10-16:
Yes, the original combination is correct in a game, but when I view the calibration tools in steam, they are incorrect. If I change the values to fix steam, then it breaks in the game controls. So the game sees it properly, but steam does not for some reason... (specifically I am using SteamFork OS)

uejji, 2024-10-16:
Is this only with the Steam Deck driver or is it also occurring with the DS5 Edge driver?

uejji, 2024-10-16:
After asking around and doing some of my own testing, if the in-game performance is as you expect, then it's probably just a visual bug on the calibration screen right now.

dansl, 2024-10-16:
I was testing with the Steam Deck driver, but if I switch to DS-edge it is now the opposite. Steam reports correct functionality, but the game is not correct...

uejji, 2024-10-23:
Moving forward, the intended result will be that gyro output matches Steam Deck.  You may have to modify your game settings if results are not expected.  See #220

closed 2025-02-25 by pastaq