ShadowBlip/InputPlumber #219: Setting Dualsense multiple times registers multiple controllers

aarron-lee, 2024-10-18:
running the following command multiple times registers duplicate controllers:

`busctl call org.shadowblip.InputPlumber /org/shadowblip/InputPlumber/CompositeDevice0 org.shadowblip.Input.CompositeDevice SetTargetDevices "as" 3 ds5 keyboard mouse`

As far as I can tell, only ds5 is affected, couldn't replicate for other controller types.

replicated on:

- SteamFork
- Bazzite

[image]

pastaq, 2024-10-18:
From @uejji :

> might've found an input plumber bug. could you test this real quick? run it multiple times, it'll register multiple ds5 controllers
```
busctl call org.shadowblip.InputPlumber /org/shadowblip/InputPlumber/CompositeDevice0 org.shadowblip.Input.CompositeDevice SetTargetDevices "as" 3 ds5 keyboard mouse
```

uejji, 2024-10-18:
> From @uejji :
>
>
> ```
> busctl call org.shadowblip.InputPlumber /org/shadowblip/InputPlumber/CompositeDevice0 org.shadowblip.Input.CompositeDevice SetTargetDevices "as" 3 ds5 keyboard mouse
> ```

This looks familiar!

pastaq, 2025-02-25:
Is this still occuring in the latest version?