ShadowBlip/InputPlumber #222: OpenGamePadUi ignores Gpd Win Max 2(2003) gamepad buttons when InputPlumber service is running

j3ffrw, 2024-11-03:
good day,
 I just updated to ogpui 0.34.1 and installed InputPlumber 0.36.0 as its requirement.
 But when I start it(inputplumber) via systemctl ogpui ignores my gpd win max 2 gamepad buttons, in Controllers menu it says "no gamepad to configure". 
When I stop the InputPlumber, gamepad button seems ok within ogpui, only thing is in Controllers page it says "InputPlumber service not available" 
Am I setting up ogpui/inputplumber correctly? since docs says it should be started. 

I'm running Nobara 40 btw, and I run opengamepadui via terminal.

pastaq, 2024-11-03:
To figure out what is going on we will need a few logs to start with.

First, lets make sure our info on your system is correct. GPD likes to reuse data in their refreshes while changing things so it can be difficult to keep everything organized for them.

Run this command to gather your DMI data
`for f in /sys/class/dmi/id/*; do echo "$f:"; sudo cat $f; done > dmi.txt`
upload the generated dmi.txt here

Run this command to grab all the input devices.
`cat /proc/bus/input/devices > input.txt`
upload the generated input.txt here

Finally, run inputplumber in debug mode, use `ctrl+c` to quit after you press some of the non-working buttons
```
sudo systemctl stop inputplumber
sudo LOG_LEVEL=debug inputplumber 2>&1 | tee inputplumber.log
```
upload the generated inputplumber.log here

Thanks

j3ffrw, 2024-11-03:
hi @pastaq , thanks for the quick response. here's the requested infos
[inputplumber.log](https://github.com/user-attachments/files/17609401/inputplumber.log)
[input.txt](https://github.com/user-attachments/files/17609402/input.txt)
[dmi.txt](https://github.com/user-attachments/files/17609403/dmi.txt)

pastaq, 2024-11-03:
Looks like the config file has the wrong phys path for the input devices, should be an easy fix to add the updated ones.

For later:
I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
N: Name="  Mouse for Windows"
P: Phys=usb-0000:65:00.3-4/input0

I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
N: Name="  Mouse for Windows"
P: Phys=usb-0000:65:00.3-4/input1

I: Bus=0003 Vendor=045e Product=028e Version=0101
N: Name="Microsoft X-Box 360 pad"
P: Phys=usb-0000:65:00.3-3/input0

pastaq, 2025-02-25:
Is this resolved in #283?