ShadowBlip/InputPlumber #224: [bug] Rumble translation for DS5 controller doesn't work as expected

imLinguin, 2024-11-06:
In the case of Hades from Epic Games Store. The motors of the internal Steam Deck controller running for short duration.  

Same events for DS5 controller result in motors vibrating for fixed amount of time, 
subsequent events cause controller to chain those events causing very long rumble

strycore, 2025-02-20:
I am experiencing this on pretty much every game (Bloostained, Bioshock 2, Dead Island 2 and more) played with the DualSense where the vibration goes on for several seconds.

pastaq, 2025-02-20:
The DS5 events are translated to Uinput for most physical devices, with actual DS5 and things like the steam deck controller getting translation to the proprietary hidraw implementation. Can you post the source devices as well so we can pinpoint where improvement needs to happen?

closed 2026-10-10 by pastaq