ShadowBlip/InputPlumber #242: Emulate PS/2 mouse sample rates

QmwJlHuSg9pa, 2024-12-13:
Hi, is it possible to downsample input sample rates, so that it seems as though a lower sample rate is being utilised e.g. Downsampling a 1000 hz USB mouse input to 40 hz (standard PS/2 mouse sample rate)?

This could be useful for older games that expect a low sample rate for consistent mouse input. E.g. both Army Men 2 and Star Wars: Shadows of the Empire (both via Wine) come to mind. 125 hz (USB minimum) is still too high for these games -- at higher sample rates, purely horizontal mouse input also results in unintended vertical movement.

Windows 98 apparently defaulted to a sample rate of `40`, before [Windows 2000](https://web.archive.org/web/19991013075639/http://www.microsoft.com:80/hwdev/input/drv.htm) upped the default to `60`.

Valid values are given [here](http://users.utcluj.ro/~baruch/sie/labor/PS2/PS-2_Mouse_Interface.htm) as: `F3h (Set Sample Rate) - The mouse responds with "acknowledge" (FAh) then reads one more byte from the host.  The mouse saves this byte as the new sample rate. After receiving the sample rate, the mouse again responds with "acknowledge" (0xFA) and resets its movement counters.  Valid sample rates are 10, 20, 40, 60, 80, 100, and 200 samples/sec.`

pastaq, 2024-12-18:
I started looking into this and it very likely will not be possible within InputPlumber. There don't appear to be any knobs we can change on a mouse device directly. I checked all the udev attributes as well as reading multiple wiki articles on the PS/2 drivers in Linux.

PS/2 mice poll rate can be set one of two ways: 
a kernel boot argument in your bootloader: 
 `psmouse.rate=40`) 
or
a driver module setting in modprode.d:
```
$ lsmod | grep -i mouse
mousedev               24576  0
```
Note that the mousedev module, which the InputPlumber mouse uses, doesn't support a rate parameter:
```
$ modinfo mousedev
filename:       /lib/modules/6.12.4-chos1-chimeraos-1/kernel/drivers/input/mousedev.ko.zst
license:        GPL
description:    Mouse (ExplorerPS/2) device interfaces
author:         Vojtech Pavlik <vojtech@ucw.cz>
srcversion:     83B28DA62FF8FF2815E1807
alias:          input:b*v*p*e*-e*1,*3,*k*110,*r*a*0,*1,*m*l*s*f*w*
alias:          input:b*v*p*e*-e*1,*3,*k*145,*r*a*0,*1,*18,*1C,*m*l*s*f*w*
alias:          input:b*v*p*e*-e*1,*3,*k*14A,*r*a*0,*1,*m*l*s*f*w*
alias:          input:b*v*p*e*-e*1,*2,*k*r*8,*a*m*l*s*f*w*
alias:          input:b*v*p*e*-e*1,*2,*k*110,*r*0,*1,*a*m*l*s*f*w*
depends:        
intree:         Y
name:           mousedev
retpoline:      Y
vermagic:       6.12.4-chos1-chimeraos-1 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        64:15:9D:53:F9:6B:C6:DF:79:D5:48:FF:76:3A:2A:B3:03:5F:0C:E6
sig_hashalgo:   sha512
signature:      30:65:02:30:47:21:8B:2D:48:E2:8D:C0:C6:EC:4C:87:06:13:DC:FF:
                47:3F:61:3D:A6:92:D0:15:14:42:1D:C9:F1:D5:6E:B9:C7:25:AB:6E:
                3F:B7:70:47:19:62:60:F2:99:3A:AB:F9:02:31:00:DE:E2:42:54:15:
                D8:9C:12:A3:17:6A:76:59:87:73:20:36:93:50:73:5B:56:64:03:EB:
                FB:98:97:28:7B:C5:53:09:66:B9:6F:B5:18:96:4A:76:06:E9:03:5E:
                69:AF:89
parm:           xres:Horizontal screen resolution (uint)
parm:           yres:Vertical screen resolution (uint)
parm:           tap_time:Tap time for touchpads in absolute mode (msecs) (uint)
```
However the psmouse module does:
```
$ modinfo psmouse
filename:       /lib/modules/6.12.4-chos1-chimeraos-1/kernel/drivers/input/mouse/psmouse.ko.zst
license:        GPL
description:    PS/2 mouse driver
author:         Vojtech Pavlik <vojtech@suse.cz>
srcversion:     DE18587C60A004621C918A7
alias:          serio:ty05pr*id*ex*
alias:          serio:ty01pr*id*ex*
depends:        serio,libps2
intree:         Y
name:           psmouse
retpoline:      Y
vermagic:       6.12.4-chos1-chimeraos-1 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        64:15:9D:53:F9:6B:C6:DF:79:D5:48:FF:76:3A:2A:B3:03:5F:0C:E6
sig_hashalgo:   sha512
signature:      30:65:02:31:00:FA:79:55:C4:B6:E9:C9:45:4C:4F:C4:0F:6B:F1:08:
                08:E4:02:2C:7A:4F:79:BF:71:11:F1:C0:AB:AD:E7:A7:43:62:8E:9A:
                E6:39:FD:45:52:02:96:67:AB:E0:6A:AC:74:02:30:47:B2:80:37:D0:
                0D:0D:53:FE:09:4A:BB:4E:9A:A0:0E:5B:D0:AC:81:10:C4:35:6A:CE:
                68:51:F0:DF:20:B5:DC:F7:60:2E:8D:04:59:FC:19:FC:F5:C8:51:CA:
                92:5B:11
parm:           elantech_smbus:Use a secondary bus for the Elantech device. (int)
parm:           synaptics_intertouch:Use a secondary bus for the Synaptics device. (int)
parm:           proto:Highest protocol extension to probe (bare, imps, exps, any). Useful for KVM switches. (proto_abbrev)
parm:           resolution:Resolution, in dpi. (uint)
parm:           rate:Report rate, in reports per second. (uint)
parm:           smartscroll:Logitech Smartscroll autorepeat, 1 = enabled (default), 0 = disabled. (bool)
parm:           a4tech_workaround:A4Tech second scroll wheel workaround, 1 = enabled, 0 = disabled (default). (bool)
parm:           resetafter:Reset device after so many bad packets (0 = never). (uint)
parm:           resync_time:How long can mouse stay idle before forcing resync (in seconds, 0 = never). (uint)
```

So if your physical PS/2 mouse uses `psmouse` you can set it:
```
$ cat /etc/modprobe.d/options
options psmouse rate=40
```

closed 2024-12-18 by pastaq

QmwJlHuSg9pa, 2026-02-16:
For posterity:

It seems this kernel module, once enabled, can change the USB polling rate for individual devices on-the-fly:

https://github.com/p0358/usb_oc-dkms

To my understanding, with this module enabled, one may be able to conditionally trigger different mouse poll rates via a game launch script, as low as 62.5 or 31.25 Hz.

pastaq, 2026-02-16:
Hmm, interesting. I'll look into it. Thanks.