ShadowBlip/InputPlumber #243: [bug] Rumble Vibration Feedback on Ayaneo Next Lite

LukeShortCloud, 2024-12-16:
I'm not sure if this is directly an InputPlumber issue or not, but posting here for greater visibility.

"""
I Tested Tekken 7 and Street Fighter 6 and the Rumble is pretty bad.
If you hit the button slowly it gives you feedback but if you are pressing multiple buttons or a single button faster than certain speed it will stop vibrating completely.
"""

Originally posted here: https://github.com/playtron-os/playtron-os/issues/158

pastaq, 2024-12-16:
Does this happen if InputPlumber isn't running?

pastaq, 2025-02-25:
@LukeShortCloud, bump for visibility.

pastaq, 2025-03-03:
I have a Next Lite, but I'm a bit overdrawn ATM with other efforts. If they can't get the info I should be able to in a couple of weeks.

angelwoody, 2025-04-16:
I have an Ayaneo Next Lite, I installed Steamos all the buttons were working except for the start and control ones. Once I went to konsole and put the sudo words everything is not working right now. Can someone help me .? Thanks in advance

pastaq, 2025-04-17:
> I have an Ayaneo Next Lite, I installed Steamos all the buttons were working except for the start and control ones. Once I went to konsole and put the sudo words everything is not working right now. Can someone help me .? Thanks in advance

Please don't spam multiple issues.