ShadowBlip/InputPlumber #250: [BUG] MSI-Claw System/Menu buttons reversed.

zezba9000, 2025-01-07:
Hey, I'm one of the people who helped get some MSI-Claw Gamepad stuff working.

Testing InputPlumber on CachyOS I noticed menu buttons are reversed. F15 and F16 need to flip here: https://github.com/ShadowBlip/InputPlumber/blob/90d65b2700084b15ac8ad16ca585c114af105154/rootfs/usr/share/inputplumber/capability_maps/msiclaw_type1.yaml#L24

The Linux input codes look correct: https://github.com/torvalds/linux/blob/fbfd64d25c7af3b8695201ebc85efe90be28c5a3/include/uapi/linux/input-event-codes.h#L266

I've been testing this stuff with my CachyOS install script for MSI-Claw here: https://github.com/zezba9000/MSI-Claw-Gamepad-Mode/blob/main/install.sh

zezba9000, 2025-01-07:
Pull Request: https://github.com/ShadowBlip/InputPlumber/pull/251

ShadowApex, 2025-01-16:
Fixed in merged PR #251

closed 2025-01-16 by ShadowApex