ShadowBlip/InputPlumber #252: Couln't open quick menu(ctrl+f2) when Inputplumber is running

j3ffrw, 2025-01-11:
hi, when running InputPlumber I can't seem to open "Quick Menu" using either Ctrl+F2 or Menu+A. 

opengamepadui v0.35.7
InputPlumber v0.40.0
GPD win max 2 2023(with fix from https://github.com/ShadowBlip/InputPlumber/issues/222)
NobaraOS 41 

**opengamepadui log w/o inputplumber**
```
234933 [INFO] [LaunchManager] <null>:<null>: Loading gamepad profile: user://data/gamepad/profiles/global.json
234672 [ERROR] [Core] core/src/input/inputplumber.rs:175: Failed to get managed objects: ServiceUnknown("The name is not activatable")
235264 [INFO] [FocusGroup] <null>:<null>: Viewport grabbing focus on node: QuickSettingsCard
235264 [INFO] [PopupStateMachine] <null>:<null>: Switched from state <null> to quick_bar_menu
235264 [INFO] [PopupStateMachine] <null>:<null>: Stack: quick_bar_menu
235264 [INFO] [GlobalStateMachine] <null>:<null>: Switched from state menu to popup
235264 [INFO] [GlobalStateMachine] <null>:<null>: Stack: menu-> popup
```

**opengamepadui log with inputplumber**
```
15216 [INFO] [LaunchManager] :0: Loading gamepad profile: user://data/gamepad/profiles/global.json
15281 [DEBUG] [InputPlumber] :0: Loading Profile: user://data/gamepad/profiles/global.json
15281 [DEBUG] [InputPlumber] :0: Mapped Capabilities (before): {"kind":"DeviceProfile","mapping":[{"name":"Gamepad:Button:Guide","source_event":{"gamepad":{"button":"Guide"}},"target_events":[{"gamepad":{"button":"Guide"}}]}],"name":"","version":1}
15281 [DEBUG] [InputPlumber] :0: Target device needs no modifications: xb360
15281 [DEBUG] [InputPlumber] :0: Mapped Capabilities (after): {"kind":"DeviceProfile","mapping":[{"name":"Gamepad:Button:Guide","source_event":{"gamepad":{"button":"Guide"}},"target_events":[{"gamepad":{"button":"Guide"}}]}],"name":"","version":1}
```

Any suggestion I could try to fix it? Thanks

pastaq, 2025-01-11:
Can you provide a debug log for InputPlumber? 
It is best to do this from an SSH, though it will work from a console TTY as well. `ctrl+alt+f3`

- First, stop the service 
`sudo systemctl stop inputplumber`

- Then run InputPlumber manually 
`sudo LOG_LEVEL=debug inputplumber 2>&1 | tee . inputplumber.log`

- Switch back to your normal tty running OpenGamepadUI using the same action as above (usually f2 or f7, not sure on nobara so cycle through)

- Do the action 
- switch back to tty3 and `ctrl+c` to stop it 
- upload inputplumber.log here

Thanks

j3ffrw, 2025-01-12:
hi @pastaq, thanks for the reply, I notice ctrl+f2 does work with a few quirks though, 
* Open the opengamepadui "main menu/home" by pressing menu button
* press F2 several times then hold F2 for 1 or 2 seconds or more.
* press/tap CTRL button, need to release the press/tap quickly or else "quickmenu" on the left side will retract/disappear back again  

here's the logs for inputplumber, i've included opengamepaui as well in case needed.

[inputplumber.log](https://github.com/user-attachments/files/18388010/inputplumber.log)
[opengamepadui.log](https://github.com/user-attachments/files/18388009/opengamepadui.log)

with your suggestion on the other ticket this is how I added config for gpd win max 2023, I still have to compare this with my other winmax2002 to see if would be enough to distinguish the different variants. 
```
matches:
  - dmi_data:
      product_name: G1619-04
      sys_vendor: GPD
      product_sku: 7896559068635

# One or more source devices to combine into a single virtual device. The events
# from these devices will be watched and translated according to the key map.
source_devices:
  - group: gamepad
    evdev:
      name: "Microsoft X-Box 360 pad"
      phys_path: usb-0000:65:00.3-3/input0
      handler: event*
  - group: keyboard
    evdev:
      name: "Mouse for Windows"
      phys_path: usb-0000:65:00.3-4/input0
      handler: event*
  - group: keyboard
    evdev:
      name: "Mouse for Windows"
      phys_path: usb-0000:65:00.3-4/input1
      handler: event*
  - group: imu
    iio:
      name: i2c-BMI0160:00
```

pastaq, 2025-02-25:
Is this resolved in #283 ?

j3ffrw, 2025-02-25:
hi Pastaq, unfortunately no.