ShadowBlip/InputPlumber #255: GPD Win Mini: Rumble stops working after suspend

justinweiss, 2025-01-18:
After suspend and resume, the rumble stops working. Here are the logs starting after resume:

```
Jan 18 13:25:37 chimeraos inputplumber[10421]: [2025-01-18T21:25:37Z INFO  inputplumber::input::manager] Found missing device, adding source device "evdev://event9" to existing composite device: "/org/shadowblip/InputPlumber/CompositeDevice0"
Jan 18 13:25:37 chimeraos inputplumber[10421]: [2025-01-18T21:25:37Z ERROR inputplumber::input::composite_device] Failed running device: "error occurred running device"
Jan 18 13:25:37 chimeraos inputplumber[10421]: [2025-01-18T21:25:37Z INFO  inputplumber::input::manager] Preparing for system resume
Jan 18 13:25:37 chimeraos inputplumber[10421]: [2025-01-18T21:25:37Z INFO  inputplumber::input::composite_device] Preparing for system resume for: /org/shadowblip/InputPlumber/CompositeDevice0
Jan 18 13:25:37 chimeraos inputplumber[10421]: [2025-01-18T21:25:37Z INFO  inputplumber::input::composite_device] Restoring target devices: ["touchpad", "xb360", "mouse", "keyboard"]
Jan 18 13:25:37 chimeraos inputplumber[10421]: [2025-01-18T21:25:37Z INFO  inputplumber::input::manager] Finished preparing for system resume
Jan 18 13:25:38 chimeraos inputplumber[10421]: [2025-01-18T21:25:38Z INFO  inputplumber::input::composite_device] Setting target devices: ["xb360", "keyboard", "mouse", "touchpad"]
```

pastaq, 2025-02-25:
Is this still occuring in the latest version? In #284 we started refreshing all target devices after suspend.

justinweiss, 2025-03-08:
Yeah, this is still happening for me in version 0.48.0.

justinweiss, 2025-04-12:
Just another update here -- The same problem still occurs on official SteamOS w/ inputplumber 0.52.1.

PR #360 (2025-05-12): fix(CompositeDevice): Free force feedback effects on device remove
If a source device is removed and later added back, the force feedback effect map can get out of sync with the device state. The removed device's state needs to be deleted from the objects referencing it.

Fixes #255