ShadowBlip/InputPlumber #262: [Feature/Bug] IMU data is not normalized

pastaq, 2025-01-22:
Currently, all axis data is normalized either signed (-1.0 to 1.0) or unsigned (0.0 - 1.0) except for IMU data, which has custom scales applied in source/target devices to make them feel somewhat close. Instead of doing this, we should refactor all IMU implementations to normalize/denormalize the values so that any source device feels consistent on any target device.

RangerMauve, 2025-01-30:
This normalization is useful for enabling features like a gyro mouse or a gyro joystick.

A big part of the translation from 3D orientation to 2D vector would be choosing the plane along which you'd project your values. e.g. how 3DS aiming worked in Ocarina of Time or the Google Daydream controller.

The mechanic I see a lot is having a "reset orientation" trigger which sets your current position as "forward". Is there a way to specify an action to toggle the gyro mapping that could work for this and a place to store the "forward" vector that persists between translations?

[This](https://github.com/ShadowBlip/InputPlumber/blob/e4131c55b8e1cbbfa931cfab70cb18ea6c97cde7/src/input/event/value.rs#L248) is where the translation would be done.

We should detect a Target of `Capability::Mouse` and have a new method called `translate_gyro_to_mouse`.

Prior art:

- [An ESP 32 based pointer](https://community.element14.com/challenges-projects/element14-presents/project-videos/w/documents/28384/creating-an-imu-based-3d-mouse-with-an-esp32-s3----episode-636)
- [A steam mapping for games](https://steamcommunity.com/sharedfiles/filedetails/?id=3266410893)

RangerMauve, 2025-01-30:
Quote from @NeroReflex while discussing in Discord:

> There are like.... 6 algorithms generally used in-games and none of them works really good for every game type sadly
> You will need to have multiple translations available. The trivial one of discarding z is not that much good
> Valve started with one well anough and is adding more because only one simply cannot make everybody happy
> If you do your searches from the perspective of a game dev you will find multiple results doing gyro-> controls with different suggested algorithms for each type of game
> For example a popular thing they do (I absolutely dislike) is using the derivative of the rotation to do the translation
> Or simply changing the sensitivity of the translation

Personally I plan to use this for navigating my screen rather than FPS so maybe we could start with an algo that works for that and go from there.

closed 2026-10-10 by pastaq