ShadowBlip/InputPlumber #271: MSI Claw A1M - Controller mode not work (Required HID command)

banpotk, 2025-01-27:
I installed Inputplumber on my current Fedora OS.
After installing the program, Inputplumber can read and display the values ​​of the MSI Claw Controller correctly.
However, the controller is still working in Desktop mode (and it works every time after booting), so I can't use it to control any game.
I checked in the Handheld Daemon program and found that it needs to send a command to change to Controller Mode by sending the HID command
0F 00 00 3C 24 01 00 00 
After sending this command, Controller buttons can work properly.
I don't know if the Inputplumber config can make it send HID commands every boot or not require external software. Is there a way or what do I need to do to make the MSI Claw buttons work in Controller Mode?
Thanks for all operations.

pastaq, 2025-01-28:
You need to install either the DKMS driver or zezba's script. I think zezba's script is more battle tested but we need more work done on the DKMS driver so it can be upstreamed.

InputPlumber design philosophy prohibits us from implementing functionality that should exist in a kernel driver. This is to avoid locking people into our solution if they prefer to make their own or use another. Since the kernel is responsible for interfacing with and creating a userspace API for hardware, we default to that.

Links are as follows:
Driver 
https://github.com/NeroReflex/hid-msi-claw-dkms

Zazba fix: 
https://github.com/zezba9000/MSI-Claw-Gamepad-Mode

banpotk, 2025-01-28:
I would like to support immutable OS like Chimera OS or Fedora Atomic because it can't install kernel drivers. If we can make enable Controller mode, it would be much easier to use.

pastaq, 2025-01-28:
The DKMS driver is being developed by ChimeraOS. It's currently doing in stable v47.

closed 2025-02-25 by pastaq