ShadowBlip/InputPlumber #290: Legion Go S may encounter OS hang when entering into suspend status

XinoAssassin, 2025-02-17:
As title describe, Legion Go S may encounter OS hang when entering into suspend status, if disable and stop the `inputplumber.service`, issue won't happen.

`OS Environment`: SteamOS highway 3.7 - 3.8
`Device`: Legion Go S

ShadowApex, 2025-02-17:
With the Steam Deck target, you need to make sure that the `inputplumber-suspend.service` is enabled to enable suspend. Can you verify that this service is enabled on your system?

XinoAssassin, 2025-02-17:
Yes, the `-suspend.service` and `-wake.service` is enabled by default.

ShadowApex, 2025-02-17:
Could you also provide this additional information:

- The `inputplumber-suspend.service` contents (e.g. `cat /usr/lib/systemd/system/inputplumber-suspend.service`)
- The inputplumber log from last boot (`journalctl -u inputplumber --boot=-1`)

Thanks!

XinoAssassin, 2025-02-17:
`-resume.service`

```
[Unit]
Description=Trigger InputPlumber resume hook
After=suspend.target sleep.target

[Service]
Type=simple
ExecStart=/usr/bin/busctl call org.shadowblip.InputPlumber /org/shadowblip/InputPlumber/Manager org.shadowblip.InputManager HookWake

[Install]
WantedBy=suspend.target sleep.target
```

`-suspend.service`
```
[Unit]
Description=Trigger InputPlumber suspend hook
Before=suspend.target sleep.target

[Service]
Type=simple
ExecStart=/usr/bin/busctl call org.shadowblip.InputPlumber /org/shadowblip/InputPlumber/Manager org.shadowblip.InputManager HookSleep

[Install]
WantedBy=suspend.target sleep.target
```

`InputPlumber service log`
```
Feb 17 12:06:41 steamdeck systemd[1]: Started InputPlumber Service.
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber] Starting InputPlumber v0.36.6
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::manager] Started hidraw device discovery thread
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::manager] Started evdev device discovery thread
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::manager] Found a matching hidraw device hidraw://hidraw5, creating CompositeDevice
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::manager] Found matching source device for: "Lenovo Legion Go S"
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::composite_device] Creating CompositeDevice with config: Lenovo Legion Go S
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::source::hidraw] Detected Legion Go S
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::composite_device] Started listening on /org/shadowblip/InputPlumber/CompositeDevice0
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::manager] Found missing device, adding source device "evdev://event5" to existing composite device: "/org/shadowblip/InputPlumber/CompositeDevice0"
Feb 17 12:06:41 steamdeck inputplumber[955]: [2025-02-17T04:06:41Z INFO  inputplumber::input::manager] Found missing device, adding source device "evdev://event10" to existing composite device: "/org/shadowblip/InputPlumber/CompositeDevice0"
Feb 17 12:06:42 steamdeck inputplumber[955]: [2025-02-17T04:06:42Z WARN  inputplumber::input::manager] Unable to create UdevDevice from /dev/input/event17 to check initialization
Feb 17 12:06:42 steamdeck inputplumber[955]: [2025-02-17T04:06:42Z WARN  inputplumber::input::manager] Unable to create UdevDevice from /dev/input/mouse2 to check initialization
Feb 17 12:06:43 steamdeck inputplumber[955]: [2025-02-17T04:06:43Z ERROR inputplumber::input::manager] Failed to remove udev dbus interface /org/shadowblip/InputPlumber/devices/source/event17: InterfaceNotFound
Feb 17 12:06:43 steamdeck inputplumber[955]: [2025-02-17T04:06:43Z ERROR inputplumber::input::manager] Failed to remove udev dbus interface /org/shadowblip/InputPlumber/devices/source/mouse2: InterfaceNotFound
Feb 17 12:06:43 steamdeck inputplumber[955]: [2025-02-17T04:06:43Z ERROR inputplumber::input::manager] Failed to remove dbus interface /org/shadowblip/InputPlumber/devices/source/event17: InterfaceNotFound
Feb 17 12:06:43 steamdeck inputplumber[955]: [2025-02-17T04:06:43Z ERROR inputplumber::input::manager] Failed to remove dbus interface /org/shadowblip/InputPlumber/devices/source/mouse2: InterfaceNotFound
Feb 17 12:14:24 steamdeck inputplumber[955]: [2025-02-17T04:14:24Z INFO  inputplumber::input::manager] Preparing for system suspend
Feb 17 12:14:24 steamdeck inputplumber[955]: [2025-02-17T04:14:24Z INFO  inputplumber::input::composite_device] Preparing for system suspend for: /org/shadowblip/InputPlumber/CompositeDevice0
Feb 17 12:14:24 steamdeck inputplumber[955]: [2025-02-17T04:14:24Z INFO  inputplumber::input::composite_device] Target devices before suspend: ["deck"]
Feb 17 12:14:24 steamdeck inputplumber[955]: [2025-02-17T04:14:24Z INFO  inputplumber::input::composite_device] Stopping target device: /org/shadowblip/InputPlumber/devices/target/gamepad0
Feb 17 14:24:16 steamdeck inputplumber[955]: [2025-02-17T06:24:16Z INFO  inputplumber::input::manager] Finished preparing for system suspend
Feb 17 14:24:16 steamdeck inputplumber[955]: [2025-02-17T06:24:16Z WARN  inputplumber::input::manager] Unable to create UdevDevice from /dev/input/event17 to check initialization
Feb 17 14:24:16 steamdeck inputplumber[955]: [2025-02-17T06:24:16Z INFO  inputplumber::input::manager] Preparing for system resume
Feb 17 14:24:16 steamdeck inputplumber[955]: [2025-02-17T06:24:16Z INFO  inputplumber::input::composite_device] Preparing for system resume for: /org/shadowblip/InputPlumber/CompositeDevice0
Feb 17 14:24:16 steamdeck inputplumber[955]: [2025-02-17T06:24:16Z INFO  inputplumber::input::composite_device] Restoring target devices: ["deck"]
Feb 17 14:24:16 steamdeck inputplumber[955]: [2025-02-17T06:24:16Z INFO  inputplumber::input::composite_device] Setting target devices: ["deck"]
Feb 17 14:24:16 steamdeck inputplumber[955]: [2025-02-17T06:24:16Z WARN  inputplumber::input::composite_device] Channel closed waiting for response from input manager
[... cut ...]
Feb 17 14:24:19 steamdeck inputplumber[955]: [2025-02-17T06:24:19Z WARN  inputplumber::input::manager] Unable to create UdevDevice from /dev/input/by-id to check initialization
Feb 17 14:25:00 steamdeck systemd[1]: Stopping InputPlumber Service...
Feb 17 14:25:01 steamdeck systemd[1]: inputplumber.service: Deactivated successfully.
Feb 17 14:25:01 steamdeck systemd[1]: Stopped InputPlumber Service.
Feb 17 14:25:01 steamdeck systemd[1]: inputplumber.service: Consumed 22.043s CPU time, 25.5M memory peak.
```

XinoAssassin, 2025-02-17:
Do you have any way to catch log when crash?

pastaq, 2025-02-17:
How do you reproduce this issue? I attempted to recreate it for 3 hours and didn't get it once.

XinoAssassin, 2025-02-17:
> How do you reproduce this issue? I attempted to recreate it for 3 hours and didn't get it once.

Very simple, just press power button under Deck UI and will trigger crash, not only me, but also my colleagues encounter the same issue.

pastaq, 2025-02-25:
This was identified as two problems 
- SteamOS was shipping an older version of InputPlumber
- hid-steam had a (now patched) bug in it's detachment sequence.

closed 2025-02-25 by pastaq