ShadowBlip/InputPlumber #306: [Question] how to remap buttons

quadraplat, 2025-03-04:
Sorry if this is the wrong place, but could not seem to find another place to ask questions.

After compiling inputplumber I would like to remap my dualsense so it uses the nintendo layout (that is swapping the face buttons), but i cannot seem to figure out how. With input plumber running, and having connected my controller, I tried commands in the readme but I get the error
`Call failed: Unknown object '/org/shadowblip/InputPlumber/CompositeDevice0'`
And I could not figure out what to do.

Currently, I assume that either I am supposed to create the CompositeDevice myself, or that inputplumber should be creating it when I connect my controller, but I cannot figure it out. Any help would be appreciated.

pastaq, 2025-03-04:
External controllers are not managed by default. You can enable it two ways:
InputPlumber CLI:
`inputplumber devices manage-all`

or dbus:
`busctl set-property org.shadowblip.InputPlumber /org/shadowblip/InputPlumber/Manager org.shadowblip.InputManager ManageAllDevices b true`

Once all devices are being managed you can set the profile 
InputPlumber CLI:
`inputplumber device 0 load-profile /usr/share/inputplumber/profiles/mouse_keyboard_wasd.yaml`

or dbus:
`busctl call org.shadowblip.InputPlumber /org/shadowblip/InputPlumber/CompositeDevice0 org.shadowblip.Input.CompositeDevice LoadProfilePath s /usr/share/inputplumber/profiles/mouse_keyboard_wasd.yaml`

quadraplat, 2025-03-06:
Thanks, that solved my issue, now to figure out if there is a way to have all this done automatically so I don't have to run a script each time i connect my controller.

closed 2025-03-06 by quadraplat