ShadowBlip/InputPlumber #317: [Bug] - Steam Deck target device lost while detaching from a USB dock during resume

matte-schwartz, 2025-03-18:
When using a device that's configured to emulate a Steam Deck controller like the Legion Go S, it's possible to lose the target device if you follow a certain set of criteria while resuming from suspend. To reproduce this issue, you'll need a dock (I used the Steam Deck dock for this), an external display, and a Bluetooth controller like an Xbox Elite 2.

1. Connect Legion Go S to a dock with at least one display output active
2. Connect a Bluetooth Xbox controller
3. Using the Xbox controller, suspend the device from the left-side menu of Steam
4. Once the device is suspended, resume it with the power button
5. While the device is still waking up, detach the USB dock

At this point, InputPlumber will fail to properly set up the target device again leaving you with basic Xbox controls via InputPlumber without any way to open the side menus. The external controller is not a factor after resume as that will still be turned off from when you suspended the device. 

Restarting the InputPlumer service sometimes works around the issue, although there have been times where I've completely lost all input except for the touchscreen and had to reboot the device to fix it.

Log of a failure with debugging verbosity: [systemreport-20250316064436.txt](https://github.com/user-attachments/files/19307626/systemreport-20250316064436.txt). Line ~12000 is where you see a resume from suspend -> failure.

matte-schwartz, 2025-03-18:
in case it's helpful:

[lsusb-dock-pressuspend.log](https://github.com/user-attachments/files/19326956/lsusb-dock-pressuspend.log)
[lsusb-undock-postsuspend.log](https://github.com/user-attachments/files/19326957/lsusb-undock-postsuspend.log)

matte-schwartz, 2025-03-26:
the external xbox controller part of this is irrelevant, as I can get it to happen following the same steps using only the Go S itself.

matte-schwartz, 2025-04-10:
I'm fairly certain this has been fixed by using the UHID target, up to you if you want to keep it open to debug vhci

pastaq, 2025-04-11:
Not sure if there is a benefit to keeping the vchi target at this point. We'll keep it open for now until we make a decision on it.

closed 2026-10-10 by pastaq