ShadowBlip/InputPlumber #335: Ayaneo next lite control not working

angelwoody, 2025-04-16:
I have an Ayaneo Next Lite, I installed Steamos all the buttons were working except for the start and control ones. Once I went to konsole and put the sudo words everything is not working right now. Can someone help me .? Thanks in advance

pastaq, 2025-04-17:
I need more information to help you. Can you more clearly describe the following:
1.) what buttons are not working specifically (use an image if you're not sure of the names, HHD button isn't a thing)
2.) what steps you have taken so far and the specific commands you used, if any
3.) how you installed InputPlumber and what version you have.

angelwoody, 2025-04-17:
1) the buttons are the one that go to the main menu and the one that controls TDP, overlay and those things. 
2) once I set everything of the steam os I went to desktop mode, went to konsole, did the steps of the sudo steps and then everything does not work. 
3) latest version on GitHub

pastaq, 2025-04-17:
>     2. once I set everything of the steam os I went to desktop mode, went to konsole, did the steps of the sudo thing and then everything does not work.

What do you mean by  `did the steps of the sudo thing`? Can you link to me what you did or paste the exact commands? Any command in Linux can be run using sudo, so saying "the sudo thing" is as ambiguous as you can get.

pastaq, 2025-04-17:
> 1. the buttons are the one that go to the main menu and the one that controls TDP, overlay and those things.

Do the top buttons (LC/RC) work?

angelwoody, 2025-04-17:
I put: sudo systemctl enable inputplumber
sudo systemctl enable inputplumber-suspend
sudo systemctl start inputplumber

Before that all buttons work except the button that take me to main menu and the ones that controlled TDP, overlay and those things. Once I did the steps of the input plumber neither of the buttons worked.

pastaq, 2025-04-17:
Do you have any conflicting programs installed like HHD? What is your OS?

angelwoody, 2025-04-17:
Nope, before the ayaneo have Bazzite working without problems. I choose to installed input plumber after I installed steam os the newest version when those two buttons doesn’t work. Once I did the steps in konsole, neither of the buttons of my handheld work.

angelwoody, 2025-04-17:
I follow a video on YouTube when a person did the same exact steps, but on a ROG ally and worked.

asfdjl239sdk, 2025-08-15:
what was the video lol? I am trying to get my LC and RC buttons working...

asfdjl239sdk, 2025-08-21:
Well I was trying to create my own capmap but I see the latest version 0.62 added support for the LC and RC buttons. I'm not sure if they works or not in steamOS since I am having trouble getting my games to default to the right controller on the Ayaneo 3 that has the extra button support enable that allows you to map RC and LC.. Steam keeps defaulting to the regular Xbox controller which doesn't have extra buttons.

pastaq, 2025-08-21:
> Well I was trying to create my own capmap but I see the latest version 0.62 added support for the LC and RC buttons. I'm not sure if they works or not in steamOS since I am having trouble getting my [...]

This is a separate unrelated issue to the next lite buttons. Can you please open a new issue with details about your specific case?