ShadowBlip/InputPlumber #363: how to remap xbox share+X buttons to keyboard CTRL+ALT+X

amigthea, 2025-05-13:
hi, awesome project, I'm looking for some help

I cannot find the share button map here
https://github.com/ShadowBlip/InputPlumber/blob/main/rootfs/usr/share/inputplumber/schema/capability_map_v1.json

and from the readme I cannot get how to map a button combination to a key combination, is there some documentation I can look up?

thank you for the patience

pastaq, 2025-09-15:
@amigthea everything @ShadowApex said is correct, I would just caution you on mapping any regularly used buttons (i.e. X). One of the fundamental things we need to do to translate input is prevent fall-through of events that should not be processed, as they could conflict. Typically this means that if a button is part of a `macro` none of the events from that button will be emitted until the button is released no no macro match was found. This could negatively affect gameplay if you need to hold X for any reason, as the event will only trigger briefly on release.

amigthea, 2025-09-16:
that was a brilliant reading, thank you so much for the effort you both