ShadowBlip/InputPlumber #427: [Original Legion Go]  Joystick Gyro not working

r1di, 2025-08-20:
No matter what i try i cant get the gyro of the controllers working on SteamOS, is it maybe not implemented yet?
The gyro of the tablet/console works tho.

Im on steamos preview version, tried stable too, no success..

pastaq, 2025-08-21:
This hasn't been implemented yet. A driver is in progress that will expose the HID command needed to enable the high definition gyro data. Once that is done there will be some coordination with Valve to add a settings menu for the Go controller that will allow users to configure which gyro is used when connected or disconnected.

Deathbit5, 2026-03-08:
I have the latest firmware installed on both controllers available through legion space, also it was working for some time until it didnt.

Ive tried resetting the controllers multiple times but no luck, and tried running the latest version of input plumber on cachyOs with the same results. Hhd seems to fix my gyro issues, but id rather work with input plumber, ive also tried using decky plumber to change the emulated controller and this does not fix gyro even when emulating a Dual sense.

Any recommendations of what to try or where to look? At the moment when using inputplumber, gyro is completely dead

closed 2026-10-10 by pastaq